Livox-SDK/livox_high_precision_mapping #1: How to get rosbag output file to create a georeferenced las file (Not issue)?

ns-1m, 2020-05-10:
Wondering how to get rosbag output file to create georeferenced las file?

Using LivoxViewer?

Livox-SDK, 2020-05-29:
Hi ns-1m,

We will support the las file output later. Now we support the pcd file output and you can do the file transform.

Livox Dev Team

ns-1m, 2020-06-02:
Is the output "pcd file output" georeferenced?

Any instruction how to get / "transform" the "pcd file output" to georeferenced las file.

What is the software we have to use?

Livox-SDK, 2020-06-02:
Hi ns-1m,

The pcd file only include the point cloud information.

Due to some regulations, we temporarily do not provide raw data(include geographic information).

We will support the las file output in next version, if you want to get georeferenced las file now you need to save the geographic information and adapt las format in the mapping code.

Livox Dev Team

ns-1m, 2020-06-02:
What do you meant on "geographic information"? You meant projection or coordinate system?

What is the output of gnss / imu file in this system?

How do you georeferenced your las file? What file you need to get the las georeference?

If we don't get the georefencing information from APX-15 then, we are just wasting the USD 10K for the APX-15 direct referencing capabilities.

Krishnateja244, 2021-05-15:
Hii, 
I am new to this field of pointclouds, using this sensor setup on UAV for conducting survey in forest. As far from my understanding this package takes timesynchronized data from INS,GPS and  lidar inorder to georeference the pointcloud and registration of pointclouds is also done in the process. Finally it outputs a map with georeferenced pointcloud. Please correct me if my understanding is wrong. 

If my understanding is correct, how can I get a georeferenced pointcloud in .las format? 

Thanks,
Krishna.

Livox-SDK, 2021-06-17:
Hi Krishnateja244, 
You can save .las file by using Livox_Viewer

GigaFlopsis, 2021-08-21:
Hello. Have you managed to find solutions for linking data to geographical coordinate systems?