Livox-SDK/livox_high_precision_mapping #3: Affordable OpenRTK330 GNSS/IMU integrated module as alternative to APX-15 (Not Issue)

ns-1m, 2020-07-02:
Hi,

Livox_high_precision_mapping is very good FOSS for Livox. However, APX-15 board and the Pospac software are very expensive (more than USD $10,000 just for the APX-15 + PosPac software $ xx,xxxx). 

Affordable OpenRTK330 GNSS/IMU integrated module as alternative to APX-15

ACEINNA’s OpenRTK330L integrates a triple-band RTK/GNSS receiver with redundant inertial sensor arrays to provide cm-level accuracy, enhanced reliability, and superior performance during GNSS outages.

ACEINNA Ships $150 High Precision Navigation Solution for Autonomous Vehicles 
New ACEINNA OPENRTK Module Rivals Performance of $10,000 Systems
Highly Accurate, OPENRTK Module is extremely reliable, smaller, easier to integrate, and far less expensive

https://www.aceinna.com/newsDetail?id=5e84279f96c44d3f88b0ee2c

https://www.aceinna.com/newsDetail?id=5ee82f713c099038d8e4b79e

OpenRTK Developer Manual
https://openrtk.readthedocs.io/en/latest/setup-OpenRTK.html

Hope Livox will consider using OpenRTK330 GNSS/IMU integrated module as alternative to APX-15.

Thanks.

Noli

liang-tang, 2020-07-05:
I'm interested in it, coule you give me one to test?

ns-1m, 2020-07-06:
@liang-tang , I am not connected to AceInna. 

I don't know who is selling the kit. Search the net where you can get one.

liang-tang, 2020-07-07:
@ns-1m，thanks, i'll think about it.

easyrtk, 2020-09-22:
The OpenRTK330LI EVK can be purchased from Aceinna's website (https://www.aceinna.com/inertial-systems/OpenRTK330). If you do not get any response, please let me know. I can find someone to contact you.

onygal, 2020-10-15:
> The OpenRTK330LI EVK can be purchased from Aceinna's website (https://www.aceinna.com/inertial-systems/OpenRTK330). If you do not get any response, please let me know. I can find someone to contact you.

I sent them an email but they did not reply, could you help my e-mail is onias.magdalena@gmail.com

africker, 2021-03-15:
I am also interested in this thread.  Thank you @ns-1m !  Do you know if the ACEINNA software solution for autonomous vehicles would be suitable for airborne collections too?   I'll contact them directly too, but it looks like their applications are mostly ground-based from their website.  Very interested to know more...

schoeller, 2022-02-10:
Has anybody tried combining one GNSS unit (e.g. Reach RS2) and one INS unit (unknown) to assemble PPS (Reachs RS2), GNRMC (Reach RS2) and GSOF (unknown unit)? Kind regards Sebastian

kschoelpple, 2022-12-28:
Hi,
I'm also interested using the OpenRTK-330.
Any progress?

leobetancor, 2022-12-31:
I purchase the Livox MD70 and I just received my OpenRTK330. I got the MD70 to work with the basic Livox_mapping and also got the openrtk working by itself. I'm experimenting on how to get both to work together. Any help will be appreciated! Also I'll keep y'all updated on anything I find out.