Livox-SDK/livox_high_precision_mapping #5: Livox Viewer安装包无法在Ubuntu16.04下安装

MrJianglc, 2020-08-06:
大疆您好，
我使用Manifold 2-G，系统是Ubuntu16.04，在[https://www.livoxtech.com/downloads](url)下载了Livox Viewer 0.8.0(64bit) - Ubuntu 16.04，解压后按照说明文档安装，出现报错：“line 11: ./so/livox_viewer: cannot execute binary file:Exec foramt error”。我不清楚是说明情况。我通过uname -m指令查了一下系统是aarch64架构，我不清楚我所下载的软件安装包是否适合我？若是安装包的问题，是否可以提供我需要的安装包？如果不是安装包的问题，又什么原因导致的呢？谢谢

Livox-SDK, 2020-08-07:
您好，@MrJianglc
目前 Livox Viewer Ubuntu 仅支持x86-64 架构的电脑设备，不知道 ARM 架构的设备

MrJianglc, 2020-08-11:
> 您好，@MrJianglc
> 目前 Livox Viewer Ubuntu 仅支持x86-64 架构的电脑设备，不知道 ARM 架构的设备

感谢您的回复，那我如何实现实时存储采集下来的点云数据呢？

Livox-SDK, 2020-09-07:
> > 您好，@MrJianglc
> > 目前 Livox Viewer Ubuntu 仅支持x86-64 架构的电脑设备，不知道 ARM 架构的设备
> 
> 感谢您的回复，那我如何实现实时存储采集下来的点云数据呢？

您好，可以使用[Livox ros driver](https://github.com/Livox-SDK/livox_ros_driver)来采集实时数据。

closed 2021-06-16 by Livox-SDK