Livox-SDK/livox_high_precision_mapping #11: lidar .las format output

Krishnateja244, 2021-06-03:
Hii, 
I am new to this field of pointclouds, using this sensor setup on UAV for conducting survey in forest. As far from my understanding this package takes timesynchronized data from INS,GPS and  lidar inorder to georeference the pointcloud and registration of pointclouds is also done in the process. Finally it outputs a map with georeferenced pointcloud. Please correct me if my understanding is wrong. 

If my understanding is correct, how can I get a georeferenced pointcloud in .las format? 

Thanks,
Krishna.

_Originally posted by @Krishnateja244 in https://github.com/Livox-SDK/livox_high_precision_mapping/issues/1#issuecomment-841668900_

Livox-SDK, 2021-06-17:
Hi Krishnateja244,

The las format file could be save through livox_viewer

Krishnateja244, 2021-06-17:
Hii,
Thanks for your reply, but I would like to get a georeferenced .las point cloud from this package, not from livox sensor and I looked into the cpp file, there is no process of strip alignment. Could you explain how to get a an unoverlapped pointcloud?

Livox-SDK, 2021-06-18:
Hi Krishnateja244 

For now, the C++ code is not support directly outputting the las file. Also due to the regulations reason, geographic information is not allowed to be recorded in the point cloud bag. 

Livox Dev Team