Livox-SDK/livox_high_precision_mapping #14: Lidar time error

cojeffj, 2022-03-03:
Hi,
I seem to have hit a wall in getting this package running correctly. I finally have the package successfully connecting to the apx-15 and livox mid-100. However, when I run the package, I am getting "lidar time error" messages:

[image]

I also am getting periodic serial open/success then disconnect messages (not sure if this is normal):

[image]

It does appear that it's connecting to the rs232 adapter. I have verified that the output from the apx-15 is correct and double checked settings in the apx-15 interface. I've also verified in livox viewer that the lidar units are seeing the PPS pulse and syncing.

[image]

Additionally, the rviz showing the pub_pointcloud2 topic doesn't show anything, and the package doesn't seem to be writing anything to the path specified in livox_mapping.launch. 

[image]

The rviz session showing the real time lidar from the livox ros driver displays a point cloud. 

[image]

Here is a sample of the entire output if it's helpful.

<details> 
  <summary>livox_mapping output </summary>
  tac@TAC:/media/tac/C01AD6601AD652D0/Livox_output$ roslaunch livox_mapping mapping_online.launch
... logging to /home/tac/.ros/log/f5de5b5a-9b32-11ec-becd-705812e2b47d/roslaunch-TAC-5561.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://TAC:46777/

SUMMARY
========

PARAMETERS
 * /apx15_gsof_driver/baud: 230400
 * /apx15_gsof_driver/port: /dev/TTL_to_USB
 * /cmdline_file_path: livox_test.lvx
 * /cmdline_str: 100000000000000
 * /data_src: 0
 * /enable_imu_bag: True
 * /enable_lidar_bag: True
 * /frame_id: livox_frame
 * /livox_mapping_case/map_file_path: /home/tac/livox_o...
 * /multi_topic: 0
 * /output_data_type: 0
 * /publish_freq: 10.0
 * /rosdistro: melodic
 * /rosversion: 1.14.12
 * /user_config_path: /home/tac/ws_livo...
 * /xfer_format: 0

NODES
  /
    apx15_gsof_driver (apx15/apx15_gsof_driver)
    livox_lidar_publisher (livox_ros_driver/livox_ros_driver_node)
    livox_mapping_case (livox_mapping/livox_mapping_case)
    rviz (rviz/rviz)
    rviz2 (rviz/rviz)

ROS_MASTER_URI=http://localhost:11311

process[livox_lidar_publisher-1]: started with pid [5576]
process[rviz-2]: started with pid [5577]
[ INFO] [1646341129.561544824]: Livox Ros Driver Version: 2.6.0
[ INFO] [1646341129.568539179]: Data Source is raw lidar.
[ INFO] [1646341129.569278996]: Config file : /home/tac/ws_livox/src/livox_ros_driver/config/livox_lidar_config.json
Commandline input bd:100000000000000
[... 609 lines cut ...]
[ INFO] [1646341172.657428550]: lidar time error
[ INFO] [1646341172.717993495]: lidar time error
[ INFO] [1646341172.816734082]: lidar time error
[ INFO] [1646341173.021184090]: lidar time error
[ INFO] [1646341173.044669204]: lidar time error
[ INFO] [1646341173.125552201]: lidar time error
[ INFO] [1646341173.144614634]: lidar time error
[ INFO] [1646341173.246941142]: lidar time error
[ INFO] [1646341173.344708628]: lidar time error
[ INFO] [1646341173.444606974]: lidar time error
[ INFO] [1646341173.544867098]: lidar time error
[ INFO] [1646341173.645583187]: lidar time error
[ INFO] [1646341173.752368751]: lidar time error
[ INFO] [1646341173.853637725]: lidar time error
^C[rviz2-5] killing on exit
[livox_mapping_case-4] killing on exit
[apx15_gsof_driver-3] killing on exit
[rviz-2] killing on exit
[livox_lidar_publisher-1] killing on exit
livox ros driver will exit
[livox_lidar_publisher-1] escalating to SIGTERM
shutting down processing monitor...
... shutting down processing monitor complete
done
</details>

cojeffj, 2022-03-04:
I've solved this issue. 
There were a few things wrong: 
1. I did not add the user that ROS is running under to the dialout group. This is necessary for ROS to have permission to access the /dev/ttyUSBx devices.
2. ROS doesn't seem to cooperate with device symlinks specified as udev rules. I originally set up udev rules to be able to easily differentiate between the rs232 to usb and TTL to usb devices. 
3. Make sure that "legacy talker ID" is unchecked in the APX-15 settings for COM1. When the messages start with "GPRMC" rather than "GNRMC" I don't think the APX driver is able to parse the messages.

Now that these are resolved, I'm getting normal messages:
[image]

And after a small test, I'm able to get an adjusted point cloud based on IMU pose.
[image]

closed 2022-03-04 by cojeffj

supercrazysam, 2023-10-31:
Hi @cojeffj 

 I am trying to setup GPS sync for livox lidar as well. I supplied PPS pulse and sent GPRMC, the lidar timestamp_type becomes 4 (only PPS) and stuck there. It never gets to timestamp_type 3 (GPS) as described in the livox timesync manual.

Have you encountered this situation? May I ask how did you setup your system to finally get timestamp_type to 3?